softwarecrash/Victron2MQTT #7: Victron2MQTT no data via VE.direct at Cerbo GX or direct at Victron MPPT

MarkusPVHM, 2024-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

2x blink code that means no mppt data from Cerbo GX. I took a VE.direct cable, one time connected to cerbo gx and one time directly to the mppt. Always the same.  results in the MQTT Explorer attached. Also no data at homeassistant.

### Screenshots / Fotos

[image]
[image]

### Version

0.2.4

### Relevant livejson output

```json
{"ESP_Data":{"IP":"192.168.178.157","sw_version":"1.1.3"},"Device_name":"Victron2MQTT"}
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-01-14:
Hello, This software is not designed to work with a Cerbo GX. It works, for example, with mppt 75/10 and other small mppt devices if rx/tx are set correctly in the victron device. You can use Venus OS to get and manage data from a Cerbo GX.

MarkusPVHM, 2024-01-14:
> Hi Markus, Du kannst auch auf Deutsch schreiben 😉
> 
> Was für ein Ve.Direct Kabel meinst Du, dass Du genommen hast?

das originale von Victron, nicht das kurze mitgelieferte Kabel. macht das einen Unterschied, weil anderst verdrahtet? Ich habe gerade gelesen im Cerbo GX Handbuch, dass der einen eigenen MQTT Server hat unter Port 8883? Ich bin verwirrt, was muss ich jetzt bei Victron2mqqt eintagen, meinen Mosquitto oder der Cerbo Broker?

MarkusPVHM, 2024-01-14:
> Das mitgelieferte. Cerbo nutzt soweit ich weiß eigene "hardcoded" Topics. Da wirst sehr wahrscheinlich mit dem Topic von Victron2MQTT nicht weiterkommen. Probiers erstmal mit Deinem normalen Broker und dem mitgelieferten Kabel. Dann sollte es funktionieren.

mit dem mitgeliefertem Kabel und einem MPPT klappt es. Ich habe aber 2x MPPT und einen smart shunt. Muss ich dann 3x victron 2mqtt kaufen/verwenden? oder gibt es ein Verlängerungskabel für den cerbo gx, da sind ja alle 3 Geräte angeschlossen....

MarkusPVHM, 2024-01-14:
> Ich glaube, Du hast da ein riesen Verständnisproblem.... Der Cerbo hat 3 Anschlüsse, weil der 3 MPPTs verwalten kann. Ganz banal erklärt: Der Cerbo ist der Zuhörer und die MPPTs die Sprecher. Cerbo hat halt 3 "Ohren" und die MPPTs je eine "Stimme". Wenn Du jetzt die Platine an den Cerbo (Zuhörer) anschließt, er wird Dir nicht antworten. Victron2MQTT hat nur ein "Ohr", kann also nur einem MPPT zuhören.

ok verstanden, wenn ich also in homeassistant die bei MPPT im energy modul verwenden möchte brauche ich also 2x victron2mqtt und kann die auch nicht mehr im VRM von Victron tracken. brauche ich also eine andere Lösung..... muss ich mal danach googlen,haben was von Venus OS gelesen und Nodered...  Danke für deine Infos.

softwarecrash, 2024-01-14:
Es gäbe tatsächlich die möglichkeit jeweils einen ESP in ein Kabel des mppt einzustricken als zusätzlichen zuhörer zu dem cerbo, das dürfte gehen, ist aber ne Menge Bastelei.

Es wäre evt einfacher mit einem mqtt Client den cerbo Server abzufragen bzw da die topics zu abonnieren und diese dann in HA zu verwerten, möglich ist das alles. Aber von HA hab ich auch wenig ahnung

MarkusPVHM, 2024-01-14:
> Es gäbe tatsächlich die möglichkeit jeweils einen ESP in ein Kabel des mppt einzustricken als zusätzlichen zuhörer zu dem cerbo, das dürfte gehen, ist aber ne Menge Bastelei.
>
> Es wäre evt einfacher [...]

wie wäre es mit einem VE.direct Y-Kabel? der eine Zweig original victron verkabelt, der andere Zweig für victron2mqtt angepasst? also eigentlich nicht so viel Bastelei. 2x ESP und 2x Y-kabel....

softwarecrash, 2024-01-14:
> wie wäre es mit einem VE.direct Y-Kabel? der eine Zweig original victron verkabelt, der andere Zweig für victron2mqtt angepasst? also eigentlich nicht so viel Bastelei. 2x ESP und 2x Y-kabel....

In der Theorie kann man einen wemos mit gnd und RX in das Kabel hängen zwischen mmpt und cerbo, ggf muss man noch Dioden einsetzen um die Pegel zu trennen.
Kann ich leider nicht testen da ich keinen cerbo oder mppt da habe

MarkusPVHM, 2024-01-23:
> > > Es gäbe tatsächlich die möglichkeit jeweils einen ESP in ein Kabel des mppt einzustricken als zusätzlichen zuhörer zu dem cerbo, das dürfte gehen, ist aber ne Menge Bastelei.
> > > Es wäre evt einfacher mit einem mqtt Client den cerbo Server abzufragen bzw da die topics zu abonnieren und diese dann in HA zu verwerten, möglich ist das alles. Aber von HA hab ich auch wenig ahnung
> > 
> > 
> > wie wäre es mit einem VE.direct Y-Kabel? der eine Zweig original victron verkabelt, der andere Zweig für victron2mqtt angepasst? also eigentlich nicht so viel Bastelei. 2x ESP und 2x Y-kabel....
> 
> Wenn Du das ausprobieren möchtest:
> 
> * Bekommst die Y-Kabel gebaut? Schick Dir da auch gern die Beschaltung für die Variante mit Dioden.
> * Wenn nicht: kann Dir ein paar Adapter zusammenlöten. Aber für den Selbstkostenpreis der Buchsen, Lochraster, Kabel und Versand.
> 
> Ich habe zwar einen MPPT an der Testwand für die Platinen, aber keinen Cerbo, um das selber testen zu können. Und den kaufen bei dem Preis, nur um damit testen zu können... ne, das ist preislich heftig.

Ok, ich habe jetzt ein original Victron Kabel zerschnitten:

Pin1-->schwarz
Pin2-->grün
Pin3-->weiß
Pin4--> gelb

Wenn ich von vorne auf die Platine vom D1mini schaue, ist da Pin1 links oder rechts? Je nachdem ich das mitgelieferte Kabel von victron2mqtt draufstecke, ict einmal schwarz links oder gelb links?
Ich würde das zunächst mal ohne Dioden verlöten, + mit +,ground mit ground und dann kreuzweise probieren Pin2 und Pin3. Wozu brauche ich die Dioden?

closed 2024-03-08 by github-actions[bot]