softwarecrash/Victron2MQTT #10: [Bug]:

Spawny2015, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Bei neuer Version 1.14 fehlt Save und Back in der Configuration. Man kann die Daten ändern, nur leider nicht speichern

### Screenshots / Fotos

[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{"ESP_Data":{"IP":"192.168.178.130","sw_version":"1.1.4","ESP_VCC":3.355,"Wifi_RSSI":-63},"Device_name":"Victron2MQTT","Device_model":"SmartSolar MPPT 75/15 rev2","Firmware_version_16":"161","Serial_number":"HQ2237ANFZN","Voltage":26.12,"Battery_current":0,"Panel_voltage":2.1,"Panel_power":0,"Operation_state":"Off","Tracker_operation_mode":"Off","Off_reason":"No input power","Current_error":"No error","Load_output_state":"OFF","Load_current":0,"total_kWh":19.62,"today_kWh":0.6,"Max_pow_today":264,"Yesterday_kWh":1.88,"Max_pow_yesterday":337,"Day":24}
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-03-10:
just checked, the buttons are here
[image]
please reflash the firmware

Spawny2015, 2024-03-10:
Yeah. You right. After 100 times erasing it´s working.... Tried everything... 3 diffrent browser diffrent things.... but didn´t expect that the update process via Tasmotizer incl. Erase and self-resetting is these buggy.... Thanks for your support! :-)

closed 2024-03-10 by softwarecrash

iamdevnull, 2025-03-13:
@softwarecrash after activating basic auth, the save button and the back button are missing.

I downloaded [Victron2MQTT_d1_mini_1.2.5_OTA.bin.gz](https://github.com/softwarecrash/Victron2MQTT/releases/download/V1.2.5/Victron2MQTT_d1_mini_1.2.5_OTA.bin.gz) and did  "Firmware Update" but still the buttons are missing.

Can you help please.

softwarecrash, 2025-03-13:
> [@softwarecrash](https://github.com/softwarecrash) after activating basic auth, the save button and the back button are missing.
>
> I downloaded [...]

try a earase and new flash over tasmotizer

iamdevnull, 2025-03-13:
> try a earase and new flash over tasmotizer

Thx for the quick answer, is there no other solution based on the current state. I have web access. 

The only thing which i did was to set basic auth.

Now there are two issues:

- mqqt data is not flowing anymore to broker
- save and back buttons are missing

It would be great to debug the real cause.

softwarecrash, 2025-03-13:
> Thx for the quick answer, is there no other solution based on the current state. I have web access.
>
> The only thing which i did was to set basic auth.
>
> Now there are two issues:
>
> * mqqt data is not [...]

i have no idea why the buttons gone, you can check the html code of a issue

iamdevnull, 2025-03-14:
> > > > [@softwarecrash](https://github.com/softwarecrash) after activating basic auth, the save button and the back button are missing.
> > > > I downloaded [Victron2MQTT_d1_mini_1.2.5_OTA.bin.gz](https://github.com/softwarecrash/Victron2MQTT/releases/download/V1.2.5/Victron2MQTT_d1_mini_1.2.5_OTA.bin.gz) and did "Firmware Update" but still the buttons are missing.
> > > > Can you help please.
> > > 
> > > 
> > > try a earase and new flash over tasmotizer
> > 
> > 
> > Thx for the quick answer, is there no other solution based on the current state. I have web access.
> > The only thing which i did was to set basic auth.
> > Now there are two issues:
> > 
> > * mqqt data is not flowing anymore to broker
> > * save and back buttons are missing
> > 
> > It would be great to debug the real cause.
> 
> i have no idea why the buttons gone, you can check the html code of a issue have you in the username or passwort set a % character?

as workaround "try a earase and new flash over tasmotizer" worked.

Yes i had the symbol !
Maybe you can try to debug if the code cannot handle symbols.

softwarecrash, 2025-03-14:
> as workaround "try a earase and new flash over tasmotizer" worked.
>
> Yes i had the symbol ! Maybe you can try to debug if the code cannot handle symbols.

there is a global bug with the % in the input fields, it is unfixable deepens on the webserver library. 
you need to reset the esp and enter your data again without the %

iamdevnull, 2025-03-14:
No problem Thx. 

Maybe it would be a good idea to put this information in the Readme for other user.