softwarecrash/Victron2MQTT #12: [FEATURE] <Victron Mppt Laderegler Temperatur>

Selyria, 2024-03-19:
### Is your feature request related to a problem? Please describe

Wäre es möglich das man den Temperatursensor noch mit aufnimmt dieser fehlt leider. Sonst super Software weiter so

Would it be possible to include the temperature sensor, which is unfortunately missing. Otherwise great software keep it up

### Describe the solution you'd like

Die Temperatur des Ladereglers auslesen

Read out the temperature of the charge controller

### Describe alternatives you've considered

Keine

softwarecrash, 2024-03-20:
hallo,
Der temperatursensor für die batterie sollte übertragen werden wenn angeschlossen.
ist unter T in der liste aufgeführt
https://github.com/softwarecrash/Victron2MQTT/blob/main/Protocol/VE.Direct-Protocol-3.33.pdf

für den internen sensor gibt es zumindest laut diesem protokol keine daten.
kannst du ggf. über usb-serial das log bereitstellen? evt wird der wert nur verschluckt

Selyria, 2024-03-20:
Ach so nee ging nur um die Temperatur des Gerätes selber da ich das über das Skript von der **Solaranzeige** einen wert hatte in grafana. Wäre denn vielleicht die Ampere von den Solarpanels möglich auszulesen bei ihrer Software wird ja nur der umgewandelte Ampere Strom angezeigt der zur Batterie geht.

was ist gemeint mit USB Serial log bin noch in der lernphase was dies hier alles angeht

softwarecrash, 2024-03-20:
aktuell geht nur das anzuzeigen was in der verlinkten pdf in der liste steht, in abhängigkeit was der victron an daten ausspuckt

Selyria, 2024-03-20:
Alles klar ich danke für die Auskunft sind ja nur werte dann bleiben die eben weg. Jedenfalls ist jetzt alles per mqtt dran und die Kabel weg.

closed 2024-03-20 by Selyria