softwarecrash/Victron2MQTT #15: [Bug]: Victron Remote_Control  nicht mehr verfügbar

mspcoding, 2024-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Some times (one or two times a day) the new control element "victron_off_reason" gets set to "0" and the Victron SmartSolar MPPT 150/35 turns off and stays turned off.

It seems that the MQTT service is not available for a short moment and delivers: Victron Remote_Control  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden

After that all values come back online again (see whole logbook below) but the Victron Remote_Control gets turned off:
Victron Off_reason  wechselte zu 0
01:18:13 - Vor 3 Stunden
Victron Remote_Control  ausgeschaltet
01:18:08 - Vor 3 Stunden
Victron Remote_Control  eingeschaltet
01:18:08 - Vor 3 Stunden

To overcome this failure I have implemented two automatizations to reactive the SmartSolar controller.
The first one is to log some values at time of deactivation. The second automatization is to reactivate the SmartSolar controller. 

Output of the log automatization at time of deactivation:
victron_off_reason: 0
victron_current_error: No error
victron_operation_state: Off
victron_device_connection: Connected

The second automatization reactivate the SmartSolar controller in two steps. Call "Victron Remote_Control einschalten" - wait fife seconds and call again "Victron Remote_Control einschalten". Only two times of reactivation turns the controller on again. 

### Screenshots / Fotos

[image]
[image]
[image]

### Steps To Reproduce

Logbook:
13. April 2024

Victron Off_reason  wechselte zu No input power
01:18:23 - Vor 3 Stunden
Victron Remote_Control  eingeschaltet
01:18:23 - Vor 3 Stunden
Victron Remote_Control  ausgeschaltet ausgelöst durch Automatisierung Victron Remote_Control einschalten ausgelöst durch Zustand von Victron Remote_Control
01:18:18 - Vor 3 Stunden
Victron Remote_Control  eingeschaltet ausgelöst durch Automatisierung Victron Remote_Control einschalten ausgelöst durch Zustand von Victron Remote_Control
01:18:18 - Vor 3 Stunden
Victron Off_reason  wechselte zu 0
01:18:13 - Vor 3 Stunden
Victron Remote_Control  ausgeschaltet
01:18:08 - Vor 3 Stunden
Victron Remote_Control  eingeschaltet
01:18:08 - Vor 3 Stunden
Victron Load_output_state  wechselte zu ON
01:18:08 - Vor 3 Stunden
Victron Off_reason  wechselte zu No input power
01:18:08 - Vor 3 Stunden
Victron Current_error  wechselte zu No error
01:18:08 - Vor 3 Stunden
Victron Operation_state  wechselte zu Off
01:18:08 - Vor 3 Stunden
Victron Firmware_version_16  wechselte zu 163
01:18:08 - Vor 3 Stunden
Victron Device_model  wechselte zu SmartSolar MPPT 150|35
01:18:08 - Vor 3 Stunden
Victron Serial_number  wechselte zu HQ2311PJET7
01:18:08 - Vor 3 Stunden
Victron Day  wechselte zu 36
01:18:08 - Vor 3 Stunden
Victron Tracker_operation_mode  wechselte zu Off
01:18:08 - Vor 3 Stunden
Victron Device_connection  wechselte zu Connected
01:18:08 - Vor 3 Stunden
Victron Remote_Control  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Load_output_state  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Off_reason  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Current_error  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Operation_state  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Firmware_version_16  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Device_model  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Serial_number  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Day  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Tracker_operation_mode  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Device_connection  nicht mehr verfügbar
01:18:08 - Vor 3 Stunden
Victron Day  wechselte zu 36
00:18:30 - Vor 4 Stunden

### Version

1.x.x and above

### Relevant livejson output

```json
{"ESP_Data":{"IP":"192.168.0.130","sw_version":"1.2.0","ESP_VCC":3.281,"Wifi_RSSI":-59,"Flash_Size":4194304,"Sketch_Size":506896,"Free_Sketch_Space":3665920,"Real_Flash_Size":4194304,"Free_Heap":22808,"HEAP_Fragmentation":8,"WS_Clients":1,"Free_BlockSize":21032},"Device_name":"Victron","Device_model":"SmartSolar MPPT 150|35","Device_connection":"Connected","Remote_Control_State":true,"Firmware_version_16":"163","Serial_number":"HQ2311PJET7","Voltage":52.41,"Battery_current":0,"Panel_voltage":0.31,"Panel_power":0,"Operation_state":"Off","Tracker_operation_mode":"Off","Off_reason":"No input power","Current_error":"No error","Load_output_state":"ON","total_kWh":91.97,"today_kWh":0,"Max_pow_today":0,"Yesterday_kWh":8.18,"Max_pow_yesterday":1201,"Day":36}
```

### What browsers are you seeing the problem on?

Safari

softwarecrash, 2024-04-13:
du kannst auch gern auf deutsch schreiben,

es gab bei dem V1.2.0 release ein paar fehler, lade dir bitte das release nochmal herunter und mach ein update

mspcoding, 2024-04-13:
Ok danke für die schnelle Antwort. Hab jetzt noch mal das [Victron2MQTT_d1_mini_1.2.0_OTA.bin.gz] heruntergeladen und über Settings eingespielt. Jedoch weiß ich jetzt nicht ob das Update auch geklappt hat, da die Versionsnummer gleich geblieben ist. Bitte beim nächsten Mal die Versionsnummer auf 1.2.1 erhöhen. Danke.

closed 2024-04-16 by softwarecrash