softwarecrash/Victron2MQTT #19: Daten von VE.Smart Networking

ESP-Raider, 2024-08-12:
### Is your feature request related to a problem? Please describe

Ich sehe über MQTT keine Daten, die von Geräten stammen, die über VE.Smart Networking verbunden sind.
Zum Beispiel der SmartBatterySense.

### Describe the solution you'd like

Es wäre toll, wenn die Daten von SmartBatterySense auch über MQTT übertragen werden. In der App stehen die Daten vom MPPT75/15 und dem SmartBatterySense auch zusammen auf dem Display.

### Describe alternatives you've considered

Warum ist das wichtig für mich? Mein Akku-Pack steht wettergeschützt, aber außerhalb eines Gebäudes. Ich möchte wissen, wann die Akku-Temperatur 5 Grad Celsius unterschreitet, damit das Laden und Entladen abgeschaltet werden kann.

### Additional context

Ein fettes Dankeschön für Victron2MQTT...es ist (fast) perfekt für mich!

softwarecrash, 2024-08-13:
VE.Smart is a different communication protocol that is not implementet in victron2mqtt, so its not possible to get this data at the moment

ESP-Raider, 2024-08-13:
Ooh, thanks for the quick reply. Is this implementation planned?

softwarecrash, 2024-08-15:
not at the moment, the idea laying around on the todo list, but i dont have time to realize it now

closed 2024-09-29 by github-actions[bot]