softwarecrash/Victron2MQTT #21: [FEATURE] Remote Control an IO PIN senden?

Jens869, 2024-09-01:
### Is your feature request related to a problem? Please describe

Ich habe den Victron2MQTT an einen Victron SmartShunt. Dieser hat nicht die Möglichkeit einen Ausgang oder das Laden abzuschalten. Auch die beiden Eingänge des Shunts sind nur Eingänge für Spannung messen oder ein Temp-Sensor. - Trotzdem habe ich den Schalter Remote Control. - Ist es möglich das ein- und aus-schalten auf einen freien IO-Pin des WEMOS zu legen? Dann ließe sich damit ein Relay steuern.

### Describe the solution you'd like

Mit Schalter Remote Control das Serial senden und gleichzeitig IO Pin des WEMOS schalten.

### Describe alternatives you've considered

Alternativ weiteren WEMOS mit Tasmota nutzen.

Jens869, 2024-09-01:
main.h:   #define RL_PIN 15 // D8
main.ccp: void setup() {   pinMode(RL_PIN, OUTPUT);  }
status-LED.h: digitalWrite(RL_PIN,remoteControlState);
nicht getestet

softwarecrash, 2024-09-02:
Pin 15 geht normalerweise an victron-rx diesen Pin kannst du frei verwenden wenn dein Gerät eh keinen Anschluss dafür hat

Jens869, 2024-09-02:
in main.h sind belegt:
#define MYPORT_TX 12 // D6
#define MYPORT_RX 13 // D7
#define LED_PIN 02 // D4

deshalb 
#define RL_PIN 15 // D8

die globale Bool "remoteControlState" wird vom Programm dafür verwendet.
wenn ich Zeit finde muss ich es mal compilieren und testen

softwarecrash, 2024-09-02:
du musst garnichts kompilieren, du musst nur D6 nicht mit dem shunt verbinden, und stattdessen an D6 ein relais hängen

Jens869, 2024-09-02:
fürs Relay muss ich noch  einen Transistor auf eine Platine löten
und Kabel ändern,
aber wenn ich eine LED schalten kann, kein Problem.

softwarecrash, 2024-09-02:
> Prima! habe des ergänzte Programm kompilieren. Ich kann jetzt mit den drei zusätzlichen Zeilen eine LED ein- und ausschalten, an PIN D8
> 
> D6 und D7 als serielle Schnittstelle zum SmartShunt
> 
> In meinen 2. Post stehen die Änderungen. Nur drei Zeilen

das ist keine serielle kommunikation, die ist einseitig, daher wird rx(victron) als schaltsignal benutzt, das kannst du vom shunt abmachen und als signal für dein relais benutzen, macht victron exakt so, du musst im code nichts ändern.

Jens869, 2024-09-02:
Beim Victron SmartShunt 500A gibt es zwei Klemmen.
Die sind aber nur zum Spannungsmessen oder Temperatur messen.
Wenn ich RX unterbreche, dann kann ich nix an den Shunt senden.

Ein extra PIN D8 ist mir lieber um die bool remoteControlState nach außen zu leiten.

Sonst müsste ich den RX in einen Arduino schicken und dort serial auf ein/aus ändern

softwarecrash, 2024-09-02:
> Beim Victron SmartShunt 500A gibt es zwei Klemmen. Die sind aber nur zum Spannungsmessen oder Temperatur messen. Wenn ich RX unterbreche, dann kann ich nix an den Shunt senden.
>
> Ein extra PIN D8 ist [...]

was sendest du den über den wemos an den shunt das D6 belegt ist?

Jens869, 2024-09-02:
D6 und D7 als serielle Schnittstelle zum SmartShunt
D8 der Inhalt der Variable  remoteControlState

was sonst?

Jens869, 2024-09-02:
OK
digitalWrite(MYPORT_TX, remoteControlState);
digitalWrite(RL_PIN, remoteControlState);

ist das Gleiche auf zwei unterschiedlichen Pins ... 

jetzt hab ich das Rad doch erneut erfunden ;-)

softwarecrash, 2024-09-02:
> OK digitalWrite(MYPORT_TX, remoteControlState); digitalWrite(RL_PIN, remoteControlState);
> 
> ist das Gleiche auf zwei unterschiedlichen Pins ...
> 
> jetzt hab ich das Rad doch erneut erfunden ;-)

deswegen sagte ich ja, du musst am code garnichts ändern, nur dein eines kabel vom shunt trennen und da led/transistor/arduino relais platine oder was auch immer dran tüddeln

Jens869, 2024-09-02:
Jetzt hab ich eine Forke mir gebaut, wo zwei PINs die gleiche Funktion haben.
Gut das wir mal darüber geredet haben.
Immerhin hab ich noch mal VisualStudio angeworfen und es hin bekommen,
auch wenn es für die Katz war

Jens869, 2024-09-02:
Done

closed 2024-09-02 by softwarecrash