softwarecrash/Victron2MQTT #27: Display more (all) information in JSON and at the web Interface (bluetooth MAC address, bluetooth encryption key, Yield_xx_days_ago)

BorgNumberOne, 2024-10-26:
### Is your feature request related to a problem? Please describe

Hi,

in the local net, the URL: http://victron2mqtt/livejson
displays following data:

{
  "ESP_Data": {
    "IP": "x.x.x.x",
    "sw_version": "1.2.3",
    "ESP_VCC": 3.208,
    "Wifi_RSSI": -75,
    "Flash_Size": 4194304,
    "Sketch_Size": 497840,
    "Free_Sketch_Space": 3674112,
    "Real_Flash_Size": 4194304,
    "Free_Heap": 23368,
    "HEAP_Fragmentation": 25,
    "WS_Clients": 0,
    "Free_BlockSize": 16824,
    "Runtime": 4924
  },
  "Device_name": "Victron2MQTT",
  "Current_error": "No error",
  "Device_connection": "Connected",
  "Remote_Control_State": false,
  "Load_output_state": "ON",
  "Relay": "OFF",
  "total_kWh": 155.07,
  "today_kWh": 3.63,
  "Max_pow_today": 1309,
  "Yesterday_kWh": 3.88,
  "Max_pow_yesterday": 1239,
  "Day": 47,
  "Device_model": "SmartSolar MPPT 150|60 rev2",
  "Firmware_version_24": "316FF",
  "Serial_number": "HQxxxxHUxYG",
  "Voltage": 45.7,
  "Battery_current": 0,
  "Panel_voltage": 0.66,
  "Panel_power": 0,
  "Tracker_operation_mode": "Off",
  "Operation_state": "Off",
  "Off_reason": 0
}

I also have got an external Victron bluetooth dongle, so you can see that everything -including details like:

bluetooth MAC address,
bluetooth encryption key,
Yield_xx_days_ago

...are transmitted over the wires to the bluetooth dongle and finally to the VictronConnect app.

Please could you add more (all) variables (which the VictronConnect app ist able to get) and their values to the JSON Output?

### Describe the solution you'd like

It also would be great to have:
"Yield_Today" ("today_kWh"),
"Yield_Yesterday",
"Yield_2_days_ago" ... "Yield_30_days_ago"
...in the web Interface (maybe in a bar diagramm like in daly2mqtt)

### Describe alternatives you've considered

A more detailed web interface and more(all possible) values in JSON would be great.

softwarecrash, 2024-10-26:
victron2mqtt works with the VE.Direct protocol.
the possbile values you can find [here](https://github.com/softwarecrash/Victron2MQTT/blob/main/Protocol/VE.Direct-Protocol-3.33.pdf), all values that gives the device over this protocol are showed in the livejson/mqtt/HA
whats not int hte VE.Direct protocol i cant display.
the Bluetooth connection is the VE.HEX or VE.Smart communication, it have more data. but on the data port it can only have VE.HEX, that provides less data then VE.Smart (only bluetooth)

there is nothing i can do for you

closed 2024-10-26 by softwarecrash

BorgNumberOne, 2024-10-26:
Thank you very much for your detailed and fast answer.

Well could you please add more visable values to the web Interface?

It would be great, If:

"today_kWh"
"Max_pow_today"
"Yesterday_kWh"
"Max_pow_yesterday"
"total_kWh"
"Day"

"Device_model"
"Firmware_version_24"
"Serial_number"

...could be displayed before:

Battery:
Panel:
Load out:

[image]

It also would be great, if the Battery values [V, A] and the Panel values [V, W] could be seperated in a small HTML table (2x2) or seperated with tab space or a Pipe sign.

softwarecrash, 2024-10-26:
Thats in the To-Do list, but a Bit Tricky.
So Not all devices send all Data, thats variable by device function, and values without Data are ugly. So when i Work again on the webui it will become more values

BorgNumberOne, 2024-10-26:
Thank you very much. :+1:

BorgNumberOne, 2024-10-26:
Pseudocode / pseudo javascript code: If there ist no/empty value in(side) a variable then do not display the variable at all. :)

softwarecrash, 2024-10-26:
> Pseudocode / pseudo javascript code: If there ist no/empty value in(side) a variable then do not display the variable at all. :)

Sure ;)
But it must be Made and thats need s Bit of time that i doenst have at the moment.
So i have a Big To-Do list 😁

BorgNumberOne, 2024-10-28:
Some further small things for the to do list: :)

1. replace "pro**z**ess"(ing) with "pro**c**ess"(ing) everywhere in the sourc ecode
2. keep one unique capitalization style for the variable names: **either** upper case **or** lower case (**t**oday_kWh, **M**ax_pow_today)
3. use official variable names from Victron specifications (today_kWh, total kWh --> Yield_today, Yield_total)

softwarecrash, 2024-10-28:
Feel free to correct it and make a PR, im a little Bit legssthenic ;)