softwarecrash/Victron2MQTT #35: [Bug]: Demaged VE.direct Port (no voltage / signals anymore) - the second time - Victron Smart Solar 150/60

BorgNumberOne, 2025-02-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

A few weeks ago, the Victron2MQTT, which I ordered from "all-solutions.store", stopped delivering data.
The Victron2MQTT web interface is still working, but no more data were received from our Victron Smart Solar 150/60 - VE.direct port.

Then I tried to connect a Victron bluetooth dongle to the VE.direct port and tested it.
Result: the dongle didn't work - no blue LED, no function .

After this I made a measuring cable (using the VE.direct extension cable from all-solutions.store) to measure voltages on the VE.direct connection port of our Victron Smart Solar 150/60.

The result: no voltages, no signals.

I was able to convince the dealer of our Victron Smart Solar 150/60 that the VE.direct port hadn't worked from the start.

So, I was able to send our Victron Smart Solar 150/60 back to the dealer and we received a new Victron Smart Solar 150/60.

After receiving the second new Victron Smart Solar 150/60, I first tested the Victron bluetooth dongle on the VE.direct port.
The result was: the blue LED lit up without any problem and the function was with success.

Then I connected the Victron2MQTT(with USB power supply) to the new/replaced Victron Smart Solar 150/60 and the result: after a short time there was no data in the web interface of the Victron2MQTT anymore.

Then I tried the Victron bluetooth USB dongle again and the dongle didn't work anymore again.

Subsequent measurements on the VE.direct port of the Smart Solar 150/60 again showed no voltage and no signals.

Ergo: second Smart Solar 150/60 demaged on the VE.direct connection port.

Could there be any constellation...

a) correct/wrong order - connection of USB power supply, connection of VE.direct port

b) galvanic isolation no longer works properly on the Victron2MQTT board

...which results in a demaged VE.direct port / Smart Solar 150/60 - the second time?

Regarding our battery setup:

16LiFePo cells - 48V

DalyBms

Victron Smart Solar 150/60 (charges 48V LiFePo)

Hoymiles WR (DC side supplied by 48V LiFePo)

USB power supply for Victron2MQTT

USB power supply and Hoymiles WR are connected to the same 230V path.

Kind regards.

It would be great if all-solutions.store could supply a VE.direct extension cable with built-in digital isolator (ADum1201 / ADum3160 / Si8422).

### Screenshots / Fotos

No screenshots.

### Steps To Reproduce

Described above.

### Version

1.x.x and above

### Relevant livejson output

```json
no Relevant livejson output.
```

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2025-02-09:
when you have  the victron2mqtt PCB there is allready a isolator

BorgNumberOne, 2025-02-09:
Would it be possible that the digital isolator of my Victron2MQTT is demaged anyhow so that any new further/future Victron Smart Solar its VE.direct port will also get demaged when connecting?

softwarecrash, 2025-02-10:
its rare but possible, please just wait for @all-solutions he is the designer of the PCB

closed 2025-03-27 by github-actions[bot]