softwarecrash/Victron2MQTT #41: [HELP] <mehrfache Ordner beim MQTT>

Fagott01, 2025-03-18:
### A clear and concise description of what the problem is.

Hallo Zusammen. Ich habe folgendes Problem. Bei mir werden mehrere Ordner durch VICTRON2MQTT angelegt. Auch, wenn ich diese lösche, werden die Ordner neu generiert. Es sieht dann unter Objects => Mqtt => 0 aus:
- Victron_SmartShant
- Victron_SmartShant0
- usw.
Habe ich was falsch gemacht? Wie kann ich das Problem lösen?

Fagott01, 2025-03-18:
> all-solutions
>  left a comment
> (softwarecrash/Victron2MQTT#41)
> Wie ist denn der Wert von Wifi_RSSI?
>
> So Sachen passieren eigentlich fast nur, wenn der Wemos einen eher schlechten Empfang hat.
>
>
>
> all-solutions
>  left a comment
> (softwarecrash/Victron2MQTT#41)
> Wie ist denn der Wert von Wifi_RSSI?
>
> So Sachen passieren eigentlich fast nur, wenn der Wemos einen eher schlechten Empfang hat.

Daran kann es eigentlich nicht liegen, denn die Entfernung zur Fritz!Box nur 2 Meter sind.

Fagott01, 2025-03-18:
Der Wert ist -51

Das heißt nix. Was zeigt der im MQTT als RSSI-Wert denn an?

Das heißt nix. Was zeigt der im MQTT als RSSI-Wert denn an?

fat-randy, 2025-03-18:
Hi,
ich hatte das Problem auch bei einem RSSI Wert von ~78-82. Hatte damals einen Repeater im Einsatz der über W-Lan mit dem Router verbunden war. An dem Repeater hatte sich der Wemos verbunden. Nun habe ich direkt neben dem Wemos (ca. 30cm) einen W-Lan Access Point sitzen der per LAN an den Router angebunden ist. Neuer RSSI Wert: ~60 aber keine Fehler mehr. Dein Wert ist ja noch besser mit -51, das ist komisch. Ehrlich gesagt hatte ich auch einen besseren RSSI Wert bei mir erwartet, habe ein paar Shellys in selbem Abstand zu dem AP und da ist der Wert zwischen 30-40. Daher die Frage an "all-solutions": Kann man da was am Wemos optimieren bzgl. W-Lan empfang? MfG

closed 2025-05-02 by github-actions[bot]