softwarecrash/Victron2MQTT #43: [HELP] Keine W-Lan Verbindung seit Update auf 1.3.0

fat-randy, 2025-05-05:
### A clear and concise description of what the problem is.

Hallo zusammen,

seit dem ich das Update auf Version 1.3.0 via Webgui gemacht habe. VErbindet der D1 sich nicht mehr richtig mit dem W-Lan. Er hat eine Verbindungsstärk von nur 1 Mbit laut Fritzbox.

Wenn ich ihn neu aufsetze via https://all-solutions.github.io/Flash2MQTT/?get=Victron2MQTT erzeugt er den AP mit guter W-Lan stärke. Sobald ich ihn jedoch mit meinem W-Lan verbinde passiert wieder das gleich wie zuerst beschrieben. Hier das Log:
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/20049246/esp-web-tools-logs.txt)

Jemand eine Idee?

softwarecrash, 2025-05-06:
Hallo,
das war einm bug im release, lade es dir nochmal noch und flashe es dann sollte der fehler weg sein

fat-randy, 2025-05-06:
Hi,
leider besteht das Problem immer noch.

[esp-web-tools-logs (1).txt](https://github.com/user-attachments/files/20066906/esp-web-tools-logs.1.txt)

Habe ihn über https://all-solutions.github.io/Flash2MQTT geflasht.

[image]

Mfg

softwarecrash, 2025-05-07:
probiere es jetzt nochmal, wenn es über den webflasher immer noch nicht will flashe es mal manuell

fat-randy, 2025-05-07:
Hi,
jetzt hast geklappt. (Per webflasher).
Super Danke.

closed 2025-05-07 by softwarecrash